HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #1: Source Code available?

jubudu, 2019-11-05:
Hello, 
first of all thank you for this interesting IOT-sensor design!

I would like, to use another, cheaper sensor like the MQ135.
Is it possible, that you upload the sources in order to code for my purposes?

HomeKidd, 2019-11-05:
Currently I’m not planning to add support for the MQ135 because these sensors are not as accurate as my sensors (MQ135 is a simple resistance based analog sensor)😁

But some else has [source code](https://github.com/maccoylton/esp-homekit-air-quality-sensor) for that sensor👌

closed 2019-11-05 by HomeKidd

HomeKidd, 2019-11-06:
Also I’ve forget to mention that i’m planning to add support for other, cheaper sensors (CCS811 and maybe BME680)✌️🤓