HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #2: How to show the AirQuality stars?

tobekas, 2019-11-23:
Which characteristics must be provided to show the new star-rating air quality in Eve instead of the text „Excellent“ „Good“,... (like in the 2015 Version?)?

HomeKidd, 2019-11-23:
As i remember AirQuality Service as primary Service with VOC Density Characteristic and the old Eve characteristic for Eve Room, Temperature sensor Service with Current Temp characteristic and Status Fault characteristic, Humidity sensor Service with Current RH characteristic and Status Fault. And the Eve Log Service with all de necessary Characteristics! ✌️

closed 2019-11-23 by HomeKidd

tobekas, 2019-11-25:
hm... I tried it with the HomeKit Accessory Simulator, but it just shows the old style. No stars...

HomeKidd, 2019-11-25:
Mine is working with this setup 😳

Try to add Temp Display Unit characteristic to the Temp sensor service. As i remember this is required too for the new Room's display
Also i've` E863F131` and` E863F11D` added to Log Service

tobekas, 2019-11-25:
That's it! Characteristic "Temperature Display Units" makes the difference.
Thanks!