HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #4: About PCB

vvlang, 2020-02-12:
What are the specifications of the capacitors and resistors on the PCB？

HomeKidd, 2020-02-12:
You can find everything about the needed parts at the [Wiki Page](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/wiki/PCB-Design#v2xx-pcb)! Due to the ESP using 5V and 3.3V, any capacitors will work above 5V. 🙂

closed 2020-02-12 by HomeKidd

vvlang, 2020-02-12:
Thank you！

    

    
    
    

    
        
            
                
                    
                        
                    
                    
                        
                            要磊
                            
                            
                                
                                    邮箱：ouchyao@gmail.com
                                
                        
                    
                
            
        
    

签名由 网易邮箱大师 定制

On 02/12/2020 16:14, HomeKid wrote: Closed #4.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/issues/4?email_source=notifications\u0026email_token=AKXVQLGSMJIDI6XDKHFGK7LRCOVWNA5CNFSM4KTQQSH2YY3PNVWWK3TUL52HS4DFWZEXG43VMVCXMZLOORHG65DJMZUWGYLUNFXW5KTDN5WW2ZLOORPWSZGOWSLIUVA#event-3029764692",
"url": "https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/issues/4?email_source=notifications\u0026email_token=AKXVQLGSMJIDI6XDKHFGK7LRCOVWNA5CNFSM4KTQQSH2YY3PNVWWK3TUL52HS4DFWZEXG43VMVCXMZLOORHG65DJMZUWGYLUNFXW5KTDN5WW2ZLOORPWSZGOWSLIUVA#event-3029764692",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]