HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #7: support

Tao173, 2020-05-14:
Thanks for this project. Does it support cheap air sensors now?

Tao173, 2020-05-14:
I have finished the PCB production, now I am very annoyed because the cost of the air sensor is too high, is there any air sensor that can be replaced without changing the PCB? That would be great If you can help me solve it.

HomeKidd, 2020-05-14:
Hi! 
Thanks 🙌
Sorry but i’m not planning to add support for cheaper sensor since there no option for a **good** CO2 sensor:

[BME680](https://s.click.aliexpress.com/e/_dWdKJ2V) also 10$+ and it cannot measure CO2, only Air quality index.

[CCS811](https://s.click.aliexpress.com/e/_d8RvIVJ) also 10$+, it can measure CO2, but if you restart it in bad air, it will save it as a baseline and thinks that the bad air is good. 

[MH-Z19](https://s.click.aliexpress.com/e/_d6NtLxR) sensors are the best CO2 sensors, but they are also 18-20$+ and these are as big as the whole IAQ PCB...

[MQ series gas sensors](https://s.click.aliexpress.com/e/_dTgJDY1) are cheap, but these are really bad and inaccurate, analog sensors. Also burn-in period nedeed for 48 hours in special conditions.

[IAQ Core C](https://s.click.aliexpress.com/e/_dTdKE6t) costs 18-21$ on AliExpress and **this is the only sensor that fits the PCB!**

Also the the sensor required for this project is linked at least 3 times in the Wiki. Like [here](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/wiki) where i’ve explained why i’ve used this sensor.
 Before i buy anything for a project i always check the part list and the prices and if its too  expensive for me i dont order any PCB or part. 

If you can find a better sensor, that uses I2C bus protocol and can fit the PCB size and **your budget** feel free to open a new issue🙌

closed 2020-05-14 by HomeKidd

Tao173, 2020-05-14:
Okay thank you