HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #8: After one use, doesn't work anymore plus screen not showing anything

YorkWell, 2020-05-20:
Jó reggelt!
First of all, thanks for your work. I love your ideas and resolution.

I ordered a couple of iAQ sensors and also two STH3x to try out this development.
I've tried in two Wemos D1 mini and in a NodeMCU with same result.
I flash through esptool console in MacOS.
I've changed also sensors, screen and as I said, main boards.
I've also checked resistors and wiring.

At first connection it works, I can set it up, join my wifi network and include it into my Home app but it shows "Error 1".
Anyhow it measures air quality and CO amount. However nothing is ever shown in the 0.96" SDD1306 screen and it always shows 23º and 50% air moisture. 
I've exhaled in the iAQ sensor and it really works but nothing changes temperature not air moisture.

But when I disconnect power from the build and I reconnect it back, it doesn't work anymore. The Home app sees it but it is not showing any measure no matter the sensor (the screen continues black) and the "error 1" keep showing.

Any hint on what is going on?
Köszönöm

HomeKidd, 2020-05-20:
Szia!

Could you please send me a log file from Terminal where i can see what is going on when it shows the error? 
The IAQ sensor takes 5 minutes to heat up from boot (even from rebooting), so after 5 min it have to show values, until that it reports "Unknown" air quality. 
Since the OLED / SHT3x is necessary for this to work, without the OLED or SHT3x the code maybe crashed because the SHT3x updates the OLED screen values.

The screen also has a 5 minute timer, here comes the button connected to GPIO0 and GND:  pressing it will turn on the screen and auto timer will turn off 5 minutes later to prevent the screen burn-in. 

What i suspect is some sort of problem with the SHT3x. Maybe the I2C bus address is wrong since the code only accepts address 0x044
`SHT3x_ADDR_1 0x44        // ADDR pin connected to GND/VSS (default)`

For better understanding the problem, a log file would be helpful, because it can tell whats the issue 😄 If either the screen / IAQ sensor / SHT3x has a problem, you should see something like that:

`!!! HomeKid ! ERROR SHT3X reading failed`

But it still looks a hardware issue to me because i have 3 of these running the same code, with different sensors (SHT30, SHT31, SHT35) since March with no problem.

Csak nem magyar vagy? 😄

HomeKidd, 2020-05-20:
Szia!

Could you please send me a log file from Terminal where i can see what is going on when it shows the error? 
The IAQ sensor takes 5 minutes to heat up from boot (even from rebooting), so after 5 min it have to show values, until that it reports "Unknown" air quality. 
Since the OLED / SHT3x is necessary for this to work, without the OLED or SHT3x the code maybe crashed because the SHT3x updates the OLED screen values.

The screen also has a 5 minute timer, here comes the button connected to GPIO0 and GND:  pressing it will turn on the screen and auto timer will turn off 5 minutes later to prevent the screen burn-in. 

What i suspect is some sort of problem with the SHT3x. Maybe the I2C bus address is wrong since the code only accepts address 0x044
`SHT3x_ADDR_1 0x44        // ADDR pin connected to GND/VSS (default)`

For better understanding the problem, a log file would be helpful, because it can tell whats the issue 😄 If either the screen / IAQ sensor / SHT3x has a problem, you should see something like that:

`!!! HomeKid ! ERROR SHT3X reading failed`

But it still looks an hardware issue to me because i
Csak nem magyar vagy? 😄

HomeKidd, 2020-05-21:
@YorkWell Any news? 🤓

YorkWell, 2020-05-21:
Hello,
I'm not Hungarian but I work with some of them ;)
Thank you for your quick response. 
I did check the reset button single push to activate the screen but it didn't work.
I don't know were to check that log because the faults comes on the iPhone Home app.
Today I'm going to test the SHT3x sensors with another sketch on Wemos D1 to see if they have 0x44 as address or if they work. I read about Wemos has not power enough to pull up these sensors and maybe it requiere additional pull up resistor for them.
The screen, one was brand new and the other I already used it in other project and it worked on it. They have address 0x78. Anyway I'll also test them.
I'll come to you with the results.
Cheers,
Jorge

HomeKidd, 2020-05-21:
I thought you're hungarian too 😄 

Simple just plug in the Wemos to  your USB port and run `screen /dev/<YouUSBPort> 115200` in Terminal like [here](https://apple.stackexchange.com/questions/32834/is-there-an-os-x-terminal-program-that-can-access-serial-ports) for log/debug.

Wemoses are cable of running this setup, i have a soldered jumper-wire madness for test rig and works perfectly! 😄 Also the pull-up resistors should be - by default - on the STH3x PCB 😄 

Also note that the OLED screen should have [I2C bus address 0x3C](https://github.com/SuperHouse/esp-open-rtos/blob/bc979883c27ea57e948daa813e2bca752ebd39e1/extras/ssd1306/ssd1306.h#L23), this is the default for all SSD1306 screens, and actually i've never seen 0x78 😳

Other thing to mention is I've wrote the error-checking code for indicating the problem withtin Homekit this way: 
* when IAQ sensor reading fails, it should indicate Error for the IAQ sensor Characteristic
* when SHT3x  sensor reading fails, it should indicate Error for the Temperature/Humidity Sensor Characteristic
* when both sensor reading fails, it should indicate Error for both IAQ sensor and Temperature/Humidity Sensor Characteristic too

So as I've said earlier it still looks like you have problems with the SHT3x 😄

YorkWell, 2020-05-21:
Hi again,
It thought on watching it through Arduino IDE serial monitor and this is what it shows:

>>> HomeKit: [Client 7] Pair Verify Step 2/2
>>> HomeKit: [Client 7] Found pairing with 1EAD8C1B-85F2-40DF-93E3-685DB992A745
>>> HomeKit: [Client 7] Verification successful, secure session established
>>> HomeKit: [Client 7] Get Accessories
i2C or sensor error!
i2C or sensor error!
Couldnt read data from sensor

So definitely it is something on the screen or the SHTX-DIS module.
I use this sensors:
https://es.aliexpress.com/item/4000139863432.html?spm=a2g0s.9042311.0.0.28d763c0fRhRZu
https://heltec.org/project/096-oled/

Cheers,
Jorge

HomeKidd, 2020-05-21:
I've checked my code and for the Errors you should see:

`!!! HomeKid ! ERROR iAQ Core C reading failed`
`!!! HomeKid ! ERROR SHT3X reading failed`
`!!! HomeKid ! ERROR failed to init SSD1306 screen`

I've checked my [old version](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/releases/tag/1.0.3) of IAQ code, where I've used DHT22 for temperature/humidity readings, and there you should see `Couldnt read data from sensor` error message when DHT22 reading fails. Also this code doesn't support any type of screen.

**Please check your firmware number in the Home app, you should see 1.0.x, but the [latest release](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/releases) is 2.0.2**😄

Also this problem _- the link pointed to the FW -_  in documentation  #5 was fixed in April! 😄

closed 2020-05-21 by HomeKidd

YorkWell, 2020-05-21:
Nagyon köszönöm!
Please check uploaded 202 version. It keep showing 103 (I understood you that it is normal) but I have exactly the same behavior: screen not responsive and Home app not showing after first connection (but it shows in Eve app...)

HomeKidd, 2020-05-21:
I've downloaded the latest v2.0.2 release and it shows 2.0.2 in the Home app and also in serial monitor too 😄 
[image]

Delete all the downloaded main.bin files and redownload the latest version again, maybe theres multiple main.bin files

YorkWell, 2020-05-21:
Excuse me! You are right. My command pointed to a different place were I put the 202 main.bin
I just flashed it correctly and it works awesome! Only issue that remains is that the screen model I have (both of them) shows weird graphics but I'll buy a new one. These are at least 3 years old.
Many thanks again. I'll tell my Hungarian mates about you :)