HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #9: wemos d1 mini and wemos shields

sorriso93, 2020-06-19:
Hello
I'm very interested in the project
Are these shields supported? [https://docs.wemos.cc/en/latest/d1_mini_shiled/index.html#environment-shields](https://docs.wemos.cc/en/latest/d1_mini_shiled/index.html#environment-shields)

in particular the SHT30 shield and SGP30 shield
Many thanks

HomeKidd, 2020-06-19:
Hi!

In theory the SHT30 shield can be used, but due to its design, its not very useful: the ESP/Wemos would generate heat, the SHT30 shield sits right above/below the ESP8266 and this will give you inaccurate temperature measurements. Other problem with that shield is it has a fixed I2C bus connection to `GPIO4`/`GPIO5` but this repository using other pins and it cannot be changed!

SGP30 sensor is not supported, see #7 Issue for details. Also see the schematics and more information at the [Wiki](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/wiki/Hardware-setup)!

closed 2020-06-19 by HomeKidd

HomeKidd, 2020-11-20:
[SGS30](https://s.click.aliexpress.com/e/_dUBLD6V) support coming soon as soon as I receive my test sensor and write the code ✌️