HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #12: Serial | USB TTL adapter

jonas240, 2021-02-22:
Hi,
how did you connect your USB TTL adapter to the serial of the 2.x.x PCB as there are only 4 out of 8 pins? I guessed you used jumper cables, however what ports should I connect to each other?
Best regards Jonas

HomeKidd, 2021-02-22:
Hi!

You only need 3.3V, GND, RX and TX from the serial adapter and put the ESP in [bootloder mode](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/wiki/Build-Instructions#putting-device-into-flash-mode) by pushing and holding the button before plugging the USB adapter into you computer.

[image]

closed 2021-02-22 by HomeKidd

jonas240, 2021-02-22:
The button on the PCB or should there be a button on the adapter?

HomeKidd, 2021-02-22:
The button on the PCB ✌️

jonas240, 2021-02-22:
Pushing the button; powering; ￼ releasing the button; connecting to USB port? 
Is that correct? 
Thanks for your help!✌️￼￼