HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #13: How long is the Esp8266 Download Tool supposed to sync?

jonas240, 2021-04-12:
Hi,

I Assembled￼ the PCB ￼as you described it. I tried to flash it, however it just starts to "sync" and nothing else happens. Should this take seconds or several minutes?

[image]￼
[image]
[image]

Or is something wrong configured? I thought I matched your specs. 

Another possible mistake: connecting the wires wrong? As I could find a datasheet for the adapter itself, I used the datasheet of a ESP01 to find VCC GND RX TX. 

How I set the ESP into boot loader mode: 1. connect the micro usb for power 2. Press and hold button 3. Connect usb of the adapter 4. Release button 5. Press "start" of the download tool

Thank you so much for your help!
Jonas

HomeKidd, 2021-04-13:
Hi!

Some adapters have a jumper on the PCB for selecting 5V/3.3V. Also some FTDI adapters have inverse TX/RX pins. Did you tried to flash it by changing the TX/RX pins? In normal situation for flashing you need to connect the adapters RX to the ESP's TX and vica versa RX to TX. My adapters are already inverse so i always connect TX-TX and RX-RX.

Its possible that the ESP is dead due to the 5V beacuse the TX/RX pins are only 3.3V tolerant on all ESP boards. You can try to power it via USB, the built-in blue LED on the ESP should flash once when you plug it into power, that way you can check is it fried or not. Changing the ESP on the PCB also could help if its fried.

jonas240, 2021-04-14:
The LEDs flash but The ESP Flash tool dosen't recognise the ESP anymore. I'll just start over :) some more practice in soldering is probebly useful for me.

jonas240, 2021-04-14:
Somehowe I've got a new question now: What does the two pin openings do? What am I supposed to connect to these? I marked them red in the picture. Last time I connected the USB to the area I marked green.
[image]

HomeKidd, 2021-04-15:
> I could flash a second ESP on a differned PCB! (Do I have to keep the button pressed during the whole flash or just during the "connecting to PC" part for a couple of seconds?)

You need to press the button and pressing it while powering up the USB adapter, after that you can release it. The button need to be held low ( aka pushed) when the ESP powers up ( aka you plug in the adapter to PC) because it can only enter **boot mode** in that way.

> What does the two pin openings do? 

Those are external power supply pins, the rectangle one should be +5V, the round is GND. I've added it for safety 😄 I dont have any photos of the PCB because I'm using these two pins to power it up, also on my newer/other designs i've started using USB Type-C 😄

jonas240, 2021-04-15:
Thanks for that! Do you know if it's a problem if not only the far left and far right pin of the micro USB port is soldered? Like when due to my limited skills for example the VCC and D- pin are shorted?

HomeKidd, 2021-04-15:
> VCC and D- pin are shorted?

Solder bridge between these pins can be a problem from the USB cable / wall adapter part, but on the PCB only 5V and GND are connected from the USB because of this 🤓

jonas240, 2021-04-15:
Thanks for the quick reply. I'll try my best to fix that than :)

closed 2021-04-27 by jonas240