HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #14: Powering PCB other than micro USB

jonas240, 2021-04-27:
Hi @HomeKidd, third time is a charm. 
I finally got the third PCB up and running and even got it added to HomeKit. Sadly it only worked about 1 minute. Every time the PCB is moved (very minor like trying to triple press the button to change the display type), the power cuts out. So I guessed: The micro USB port is lose. I checked again and as far as I can tell the two small pins are still connected. Still if I attach a cable and put slowly pressure on it (like pushing up or down; it actually doesn't matter) the blue LED from the ESP lights up for a very short time and the usual high pitch tone starts for a very short time and stops right away. I can not find a cable position where it is powered for a constant time like it did for the first minute. 

So you told me the two "holes" near the USB port are like a backup power supply. Can you guide me on how to use that? As I have no idea on how to fix the USB port problem (still I'm going to try a fourth time, but I'm lacking an idea what's wrong with my soldering on this). Or do you have an idea what might be the problem with my soldering here? 

Best regards
Jonas

closed 2021-04-27 by jonas240

HomeKidd, 2021-04-28:
@jonas240 

I've [updated the PCB with slightly different design](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/wiki/PCB-Design), but most importantly with USB Type-C connector!🤓

> So you told me the two "holes" near the USB port are like a backup power supply. Can you guide me on how to use that?

You have two holes on the PCB, the square one has a '+' sign next to it, that is +5V input, the onther round one is GND 😄 You can use a [microUSB adapter PCB](https://s.click.aliexpress.com/e/_A1pOWD) or an [USB Type-C adapter PCB](https://s.click.aliexpress.com/e/_9x6jsh) to connect wires to these two holes. Data pins _(D+, D-)_ don't do anything, you only need the GND and +5V connections from the adapter board 😄

jonas240, 2021-04-28:
You are amazing! I just orders 10 pieces.