HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #15: VOC density as value?

jonas240, 2021-04-27:
Hi, is it possible to see the VOC density as a value like ppb in the eve or home app? Additionally to the five stars as indicators for Air Quality.
Great work!

Best regards
Jonas

HomeKidd, 2021-04-27:
You can see it in the Home app when you open the accessory’s setting page:

[image]

closed 2021-04-27 by HomeKidd

jonas240, 2021-04-27:
Hi @HomeKidd 
This is somehow not available for me. 

[image]
[image]

HomeKidd, 2021-04-27:
The Eve app doesnt show the VOC because it should be part of the Eve history which is dont have (yet)😁
The Home app has it but german translation (like my Hungarian one😅) spells all the words of VOC, not like the english translation 😬 its the Home app’s / translation’s mistake 😬
If you set the language to US on the phone i’m 100% sure you’ll see the text and the values too 🤓
Sadly you cant do anything with that, its an Apple/Home app related problem 🤷🏻‍♂️😩