HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #16: 2.0.2 not working when 2.0.1 is

wachl, 2021-04-30:
Szia!
köszönöm szepen for this awesome project!
I can't get 2.0.2 to work, serial output is just gibberish. 2.0.1 is working fine though!
However my understanding was, that pressing the button on the pcb would activate the display, but it resets completly (at least at 2.0.2)?

Thank you for your help!

HomeKidd, 2021-05-01:
Hi!
The [latest release](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/releases/tag/2.1.0)
Is 2.1.0 which have features like turning off the reset function 30 seconds after boot🤓
Your problem is looks like to hardware issue to me: 
When you se garbage on the serial the means you flashed the ESP with wrong flash mode 

The resetting needs 10+ seconds pressing to the button maybe a sticky button? Or the wrong flash mode problem causing this too

wachl, 2021-05-01:
Hi!
thank you for your quick reply. It was late last nigth and I screwed up the version numbers in my original post (already edited).

Still my issue is that 2.0.2 is working, only the reset pin is completly resetting everything and not turning the screen on. It is not a sticky button since I just short the pins manually.

I tried again to get 2.1.0 working (this time under MacOS) but it still wont work.
I used this command to flash (ofc I erased the flash beforehand)
`esptool.py -p /dev/cu.usbserial-A50285BI --baud 115200 write_flash -fs 4MB -fm dio -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 main.bin`

And this is the serial readout after I joined the wifi and added to homekit:

`[Disconnected]
[Connected]
scandone
scandone
scandone
add 1
aid 1
station: 2c:f0:ee:25:34:8a join, AID = 1
scandone
scandone
scandone
>>> wifi_config: Connecting to FRITZ!Box 7530 TU
scandone
add 0
aid 5
cnt 

connected with FRITZ!Box 7530 TU, channel 1
dhcp client start...
ip:192.168.178.56,mask:255.255.255.0,gw:192.168.178.1
>>> wifi_config: Connected to WiFi network
station: 2c:f0:ee:25:34:8a leave, AID = 1
rm match
bcn 0
del if1
mode : sta(e8:db:84:98:4a:2c)
Function called without core lock
Connected to WiFi
HomeKid > Starting HomeKit Server
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x8c000
>>> HomeKit: Generated new accessory ID: 47:E4:1D:F8:CB:18
>>> wifi_config: Stopping HTTP server
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=AirMonitor-984A2 md= AirQuality Sensorpv=1.0id=47:E4:1D:F8:CB:18c#=8256s#=1ff=0sf=1ci=10
                                 sh=eKOMQA== Port=5556 TTL=4500
>>> wifi_config: Stopping DNS server
>>> HomeKit: Got new client connection: 3 from 192.168.178.39
>>> HomeKit: [Client 3] Pair Setup Step 1/3
>>> HomeKit: [Client 3] Pair Setup Step 2/3
>>> HomeKit: [Client 3] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 5BB68219-E557-42CB-87BD-495D99FF4531
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=AirMonitor-984A2 md= AirQuality Sensorpv=1.0id=47:E4:1D:F8:CB:18c#=8256s#=1ff=0sf=0ci=10
                                 sh=eKOMQA== Port=5556 TTL=4500
>>> HomeKit: [Client 3] Successfully paired
>>> HomeKit: [Client 3] Closing client connection
>>> HomeKit: Got new client connection: 3 from 192.168.178.39
>>> HomeKit: [Client 3] Pair Verify Step 1/2
>>> HomeKit: [Client 3] Pair Verify Step 2/2
>>> HomeKit: [Client 3] Found pairing with 5BB68219-E557-42CB-87BD-495D99FF4531
>>> HomeKit: [Client 3] Verification successful, secure session established
>>> HomeKit: [Client 3] Get Accessories
>>> HomeKit: [Client 3] Update Characteristics
>>> HomeKit: [Client 3] Update Characteristics
>>> HomeKit: [Client 3] Update Characteristics
>>> HomeKit: [Client 3] Add Pairing
>>> HomeKit: Added pairing with 2C6B201A-7CEC-4B99-9C5A-C04B75010D6D
[... 241 lines cut ...]
mode : sta(e8:db:84:98:4a:2c)
add if0
scandone
reconnect
scandone
add 0
aid 4
cnt 

connected with FRITZ!Box 7530 TU, channel 1
dhcp client start...
ip:192.168.178.56,mask:255.255.255.0,gw:192.168.178.1
>>> wifi_config: Connected to WiFi network
Connected to WiFi
HomeKid > Starting HomeKit Server
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 47:E4:1D:F8:CB:18
>>> HomeKit: Found admin pairing with 5BB68219-E557-42CB-87BD-495D99FF4531, disabling pair setup
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=AirMonitor-984A2 md= AirQuality Sensorpv=1.0id=47:E4:1D:F8:CB:18c#=8256s#=1ff=0sf=0ci=10
                                 sh=eKOMQA== Port=5556 TTL=4500`

The screen stays off the entire time (working with 2.0.2). I think it might be an I2C issue?

Any help is appreciated!

HomeKidd, 2021-05-01:
What happens when you wait after the wi-fi connection? The Screen task is created 2 sencond after the successful wi-fi connection and HomeKit running 

Until this try to dont touch the button

wachl, 2021-05-02:
Unfortunatley simply nothing happens, no screen output and no serial output. It seems as if the ESP crashed.

HomeKidd, 2021-05-02:
> It seems as if the ESP crashed.

can you send me an image of the connections how was the esp wired?

wachl, 2021-05-02:
I am using your pcb :) (mini usb version)

Lovis Wach

wachl, 2021-05-15:
Hi, I now tried using a different serial converter but I am still getting the same result. After flashing I restart the ESP (power off / power on -> no button press) and enter my wifi credentials through the webpage. The connection is established (I can see it connected to my wifi) but display stays off. Any other ideas?
Thanks!

HomeKidd, 2021-05-15:
On first setup the display should be on and it turns off after 5 minutes, so this could be a faulty display. 

Did you updated the Fw to 2.1.0?

Did you tried to triple press the button for [changing the display type](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/wiki/Setup#display-type)?

Can you send a picture of the OLED screen?

wachl, 2021-05-15:
Yes, I am using the Fw 2.1.0
I am using [this SSD1306 screen ](https://www.amazon.de/gp/product/B01L9GC470/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&th=1 ) atm which has the VCC and GND pin switched (for testing I am using cables and connect it properly ofc).
Since I also think it might have to do something with the display, I have [this SH1106 one](https://www.amazon.de/gp/product/B078J78R45/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) (with correct pin layout) coming on monday.

The display is not defective however since it is working with fw 2.0.2. I would stay on that version but it seems unstable for me.

Thanks!

wachl, 2021-05-17:
New display, same problem :(
I also can not change the display mode, as soon as i press the button, the ESP restarts. Any idea what I could try next?
Thanks!

wachl, 2021-05-17:
Any chance you could send me the source code for fw 2.1.0 and 2.0.2 for me to check?

HomeKidd, 2021-05-18:
> Any chance you could send me the source code for fw 2.1.0 and 2.0.2 for me to check?

Nope, its still secret 😄 

> Only bug on 2.0.2 i noticed is the resetting of the homekit configuration even on a short button press :(. On 2.0.2 display (only SSD1306) is working and sensor data is read.

I'm still guessing that this button is broken. All my PCBs are running 2.1.0 and never have any issue like that. The reset code only runs when you press the button for at least 10 seconds. 
Also i have a lot of safety feature:
*  `first_init` will prevent turning the display on when the SSD1306 task is not running
* `_resettable` will prevent resetting after 30 seconds from boot

```
void button_callback(button_event_t event, void* context) {
    switch (event) {
        case button_event_single_press:
            if(!first_init){
                lcd_power = !lcd_power;
                printf("HomeKid > Display %s\n", lcd_power ? "ON" : "OFF" );
                vTaskResume(ssd1306_task_handle);
                sdk_os_timer_arm(&display_timer, 300000, 0); // 300 sec = 5 min
                save_states_callback(2500);                 
            }
            break;
        case button_event_double_press:
            break;
        case button_event_tripple_press:
            _is_ssd1306 = !_is_ssd1306;
            save_states();
            printf("HomeKid > Display Type changed to %s. Please Reboot!\n", _is_ssd1306 ? "SSD1306" : "SH1106");
            sdk_system_restart();
            break;
        case button_event_long_press:
            if(_resettable) reset_configuration();
            else printf("!!! HomeKid > Reset Failed!\n Reset can be done wihtin 30 seconds from reboot. Please reboot the device and try again!\n");
            break;
        default:
            printf("Unknown button event: %d\n", event);
    }
}
```

I've made some minor changes in[ v2.1.1](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/releases/tag/2.1.1) but i'm 100% sure that the cause of this problem is a hardware (button) related because the code never lets you fully reset on single press 😄

jonas240, 2021-07-05:
I am using the new USB-C PCB and got a simiar if not the same problem with both 2.10 and 2.11.

1. The display never workes like described here. I am able to change the display type.
2. There is no sensor data pushed. Even in serial
3. One push (to supposedly wake up the display) reconnects the Wifi. I'm not sure if this is a bug or just the way you set up the serial output.

What bugs me: I had (apart form unreliable power) a working Micro USB model with 2.10. The exact same software causes these malfunktions in the new PCB for me. However, @wachl uses the old version and has these malfunctions, but mine model worked with 2.10. I cannot wrap my head around that.

Thank you @HomeKidd for your work nevertheless!!

Here is my serial output from the very first start:
```
13:16:51.049 -> scandone
13:17:01.026 -> scandone
13:17:05.791 -> >>> wifi_config: Connecting to MeinWLAN
13:17:07.531 -> scandone
13:17:08.841 -> add 0
13:17:08.841 -> aid 6
13:17:08.841 -> cnt 
13:17:08.934 -> 
13:17:08.934 -> connected with MeinWLAN, channel 11
13:17:08.934 -> dhcp client start...
13:17:08.934 -> >>> wifi_config: Connecting to MeinWLAN
13:17:08.934 -> scandone
13:17:11.047 -> scandone
13:17:11.514 -> station: ce:d2:43:fc:ea:9c join, AID = 1
13:17:15.894 -> ip:192.168.178.44,mask:255.255.255.0,gw:192.168.178.1
13:17:18.935 -> >>> wifi_config: Connected to WiFi network
13:17:18.935 -> station: ce:d2:43:fc:ea:9c leave, AID = 1
13:17:18.935 -> rm match
13:17:18.935 -> bcn 0
13:17:18.935 -> del if1
13:17:18.935 -> mode : sta(e8:db:84:e4:dc:bc)
13:17:18.935 -> Function called without core lock
13:17:19.215 -> Connected to WiFi
13:17:19.215 -> HomeKid > Starting HomeKit Server
13:17:19.215 -> >>> HomeKit: Starting server
13:17:19.215 -> >>> HomeKit: Formatting flash at 0x8c000
13:17:19.263 -> >>> HomeKit: Generated new accessory ID: AE:7A:65:76:BD:32
13:17:19.263 -> >>> wifi_config: Stopping HTTP server
13:17:19.357 -> >>> HomeKit: Generated new accessory key
13:17:19.357 -> >>> HomeKit: Configuring mDNS
13:17:19.357 -> mDNS announcement: Name=AirMonitor-E4DCB md= AirQuality Sensorpv=1.0id=AE:7A:65:76:BD:32c#=8256s#=1ff=0sf=1ci=10sh=KVUl/A== Port=5556 TTL=4500
13:17:20.530 -> >>> wifi_config: Stopping DNS server
13:17:22.363 -> >>> mdns_recv: pbuf too small
13:17:22.363 -> >>> mdns_recv: pbuf too small
13:18:22.394 -> >>> mdns_recv: pbuf too small
13:18:22.394 -> >>> mdns_recv: pbuf too small
13:18:42.798 -> >>> HomeKit: Got new client connection: 3 from 192.168.178.27
13:18:42.798 -> >>> HomeKit: [Client 3] Pair Setup Step 1/3
13:19:06.392 -> >>> HomeKit: [Client 3] Pair Setup Step 2/3
13:19:20.722 -> >>> HomeKit: [Client 3] Pair Setup Step 3/3
13:19:20.862 -> >>> HomeKit: Added pairing with 7D133A8E-67EC-4967-B0DB-62537690E51A
13:19:20.956 -> >>> HomeKit: Configuring mDNS
13:19:20.956 -> mDNS announcement: Name=AirMonitor-E4DCB md= AirQuality Sensorpv=1.0id=AE:7A:65:76:BD:32c#=8256s#=1ff=0sf=0ci=10sh=KVUl/A== Port=5556 TTL=4500
13:19:20.956 -> >>> HomeKit: [Client 3] Successfully paired
13:19:21.003 -> >>> HomeKit: [Client 3] Closing client connection
13:19:21.049 -> >>> HomeKit: Got new client connection: 3 from 192.168.178.27
13:19:21.049 -> >>> HomeKit: [Client 3] Pair Verify Step 1/2
13:19:21.424 -> >>> HomeKit: [Client 3] Pair Verify Step 2/2
13:19:21.424 -> >>> HomeKit: [Client 3] Found pairing with 7D133A8E-67EC-4967-B0DB-62537690E51A
13:19:21.566 -> >>> HomeKit: [Client 3] Verification successful, secure session established
13:19:21.613 -> >>> HomeKit: [Client 3] Get Accessories
13:19:21.846 -> >>> HomeKit: [Client 3] Update Characteristics
13:19:21.892 -> >>> HomeKit: [Client 3] Update Characteristics
13:19:21.892 -> >>> HomeKit: [Client 3] Update Characteristics
13:19:21.939 -> >>> HomeKit: [Client 3] Add Pairing
13:19:21.939 -> >>> HomeKit: Added pairing with 948266A9-7DD1-43F7-8745-24400C445D93
13:19:21.985 -> >>> HomeKit: [Client 3] Update Characteristics
13:19:22.403 -> >>> mdns_recv: pbuf too small
[... 401 lines cut ...]
13:21:41.660 -> 
13:21:41.660 ->   AirQuality Sensor's Firmware Version: 2.1.0, for Harware Revision 2.0.0, Display is SH1106.
13:21:41.706 -> 
13:21:41.706 -> >>> wifi_config: Initializing WiFi config
13:21:41.706 -> >>> wifi_config: Connecting to MeinWLAN
13:21:41.706 -> mode : sta(e8:db:84:e4:dc:bc)
13:21:41.706 -> add if0
13:21:41.801 -> scandone
13:21:43.070 -> add 0
13:21:43.070 -> aid 6
13:21:43.070 -> cnt 
13:21:43.116 -> 
13:21:43.116 -> connected with MeinWLAN, channel 11
13:21:43.116 -> dhcp client start...
13:21:44.152 -> ip:192.168.178.44,mask:255.255.255.0,gw:192.168.178.1
13:21:51.697 -> >>> wifi_config: Connected to WiFi network
13:21:51.838 -> Connected to WiFi
13:21:51.838 -> HomeKid > Starting HomeKit Server
13:21:51.838 -> >>> HomeKit: Starting server
13:21:51.838 -> >>> HomeKit: Using existing accessory ID: AE:7A:65:76:BD:32
13:21:51.838 -> >>> HomeKit: Found admin pairing with 7D133A8E-67EC-4967-B0DB-62537690E51A, disabling pair setup
13:21:51.838 -> >>> HomeKit: Configuring mDNS
13:21:51.838 -> mDNS announcement: Name=AirMonitor-E4DCB md= AirQuality Sensorpv=1.0id=AE:7A:65:76:BD:32c#=8256s#=1ff=0sf=0ci=10sh=KVUl/A== Port=5556 TTL=4500
13:22:22.319 -> >>> mdns_recv: pbuf too small
```