HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #18: PCB V2.XX

Vlr2021, 2021-07-04:
Hello!
Your work is excellent! I like very much.
I wanted to check with you on 2. XX PCB. Will there be a new firmware for this version of the board?

jonas240, 2021-07-05:
Just use the 2.XX firmware. However, I just finished the soldering on a new USB-C Type PCB, and I've run into the same problem as described here: #16 
Display isn't working and no sensor data. Wi-Fi is working though :D 

The 2.XX PCB with Micro USB worked fine (just the Micro USB Connector was a pain in the a**)

@HomeKidd Do you got a working model 2 with USB-C?

Vlr2021, 2021-07-05:
The new version of the USB-C board has changed the connections of the sensor and display to the GPO. Therefore, the new version of the board does not work.

jonas240, 2021-07-05:
@Vlr2021 Do you know if this is solvable through software? Or does the new 2.xx won't work at all? I still got some of the old 2.xx PCBs here. Than I'll focus on these.

Vlr2021, 2021-07-05:
I think there are two possible solutions here. Either rewrite the code or fix the PCB.

jonas240, 2021-08-14:
Hi @HomeKidd could you please give us an update about this thread or #19? Even if you decide to abandon this project for now :) (Still that would be sad due to the freshly purchased PCBs but nevertheless an option) Best regards!

Vlr2021, 2021-09-14:
Hello, jonas240. Are you using a new board with USB Type-C?

jonas240, 2021-09-14:
@Vlr2021 Yes the one with USB-C. My "Micro-USB version" one had some issues with cutting power from time to time (possible due to my bad solder skills). The one with USB-C "worked" for me. Apart from the stated issues :)