HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #19: 2.10 and 2.11 detect Hardware revision 1.0.0 instead of 2.0.0 for new USB-C PCB | Display failed too

jonas240, 2021-07-05:
Hi @HomeKidd 

I am using the new USB-C PCB and got a similar issue as #16 with both 2.10 and 2.11.

1. The display never works like described in #16. I am able to change the display type. But serial output states `Hardware Revision 1.0.0`? Also  `!!! HomeKid ! ERROR display_init: failed to init SSD1306 screen` even when it's set up to `SH1106` This goes on in a constant loop!
2. Some timer doesn't work
3. There is no sensor data pushed. Even in serial. HomeKit shows data (aka no data) as pictures show in #16
4. One push (to supposedly wake up the display) reconnects the Wi-Fi. I'm not sure if this is a bug or just the way you set up the serial output.

What bugs me: I had (apart from unreliable power) a working Micro USB model with 2.10. The exact same software causes these malfunctions in the new PCB for me. However, @wachl uses the old version and has these malfunctions, but mine model worked with 2.10. I cannot wrap my head around that.

Thank you, @HomeKidd, for your work nevertheless!!

Here is a part of my serial output:
```
14:00:56.446 ->   AirQuality Sensor's Firmware Version: 2.1.0, for Harware Revision 1.0.0, Display is SH1106.
14:00:56.492 -> 
14:00:56.492 -> >>> wifi_config: Initializing WiFi config
14:00:56.492 -> >>> wifi_config: Connecting to MeinWLAN
14:00:56.585 -> mode : sta(e8:db:84:e4:dc:bc)
14:00:56.585 -> add if0
14:00:56.723 -> scandone
14:00:58.958 -> reconnect
14:00:59.100 -> scandone
14:01:00.361 -> add 0
14:01:00.361 -> aid 6
14:01:00.361 -> cnt 
14:01:00.456 -> 
14:01:00.456 -> connected with MeinWLAN, channel 11
14:01:00.456 -> dhcp client start...
14:01:01.436 -> ip:192.168.178.44,mask:255.255.255.0,gw:192.168.178.1
14:01:06.582 -> >>> wifi_config: Connected to WiFi network
14:01:06.722 -> Connected to WiFi
14:01:06.722 -> HomeKid > Starting HomeKit Server
14:01:06.722 -> >>> HomeKit: Starting server
14:01:06.722 -> >>> HomeKit: Using existing accessory ID: 0B:FC:FA:B9:E5:A5
14:01:06.722 -> >>> HomeKit: Found admin pairing with 7D133A8E-67EC-4967-B0DB-62537690E51A, disabling pair setup
14:01:06.768 -> >>> HomeKit: Configuring mDNS
14:01:06.768 -> mDNS announcement: Name=AirMonitor-E4DCB md= AirQuality Sensorpv=1.0id=0B:FC:FA:B9:E5:A5c#=8256s#=1ff=0sf=0ci=10sh=sIt4gA== Port=5556 TTL=4500
14:01:08.728 -> !!! HomeKid ! ERROR display_init: failed to init SSD1306 screen
14:01:09.804 -> Timer Stop Failed
14:01:10.271 -> Timer Start Failed
14:01:10.786 -> Timer Stop Failed
14:01:11.301 -> Timer Start Failed
14:01:11.810 -> Timer Stop Failed
14:01:12.280 -> Timer Start Failed
14:01:12.422 -> 
14:01:12.422 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
14:01:12.422 -> 
14:01:12.422 -> load 0x40100000, len 2292, room 16 
14:01:12.422 -> tail 4
14:01:12.422 -> chksum 0x57
14:01:12.466 -> load 0x3ffe8000, len 772, room 4 
14:01:12.466 -> tail 0
14:01:12.466 -> chksum 0x0b
14:01:12.466 -> csum 0x0b
14:01:12.466 -> 
14:01:12.466 -> rBoot v1.4.0 - richardaburton@gmail.com
14:01:12.466 -> Flash Size:   32 Mbit
14:01:12.466 -> Flash Mode:   DIO
14:01:12.466 -> Flash Speed:  40 MHz
14:01:12.466 -> rBoot Option: Big flash
14:01:12.466 -> rBoot Option: RTC data
14:01:12.466 -> 
14:01:12.466 -> Booting rom 0.
14:01:12.700 -> pp_task_hdl : 3fff1310, prio:14, stack:512
14:01:12.700 -> pm_task_hdl : 3fff0c00, prio:1, stack:176
14:01:12.700 -> frc2_timer_task_hdl:0x3fff5790, prio:12, stack:200
14:01:12.700 -> 
14:01:12.700 -> ESP-Open-SDK ver: 0.0.1 compiled @ Nov 20 2020 12:10:06
14:01:12.747 -> phy ver: 273, pp ver: 8.3
14:01:12.747 -> 
14:01:12.747 -> HomeKid > Loading saved states
14:01:12.747 -> 
14:01:12.747 -> HomeKid > Hardware Revision v1.0.0 found

```

fubehere, 2021-11-08:
*push

I have the same issue. Getting Error "14:01:08.728 -> !!! HomeKid ! ERROR display_init: failed to init SSD1306 screen".

Using SH1306. Switched to SH1306 by pressing button 3xtimes. But the error still exists and showing the same error. 

@HomeKidd please support us. You did amazing work. Thank you <3

fubehere, 2021-11-12:
Some update from my side: I configured a second setup on a breadboard and it’s working. I don’t know why, but my every setup on pcb fails (I did already 3, because I thought I messed up something) 

it’s frustrating :/

closed 2025-06-01 by jonalbr