HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room #24: IR Remote V1.3.1

zsoltturcsan, 2024-04-09:
SZia,
Sajnos a IR Remote V1.3.1 vezérlőhöz nem találok beültetési rajzot, vásároltam pár darabot a pcbway-ről.
Segítenél?
Előre is köszönöm!

HomeKidd, 2024-04-09:
Üdv! 
Itt a teljes BOM fájl!
[BOM_ESP8266-HomeKit-IR-Remote_2024-04-09.csv](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/files/14919385/BOM_ESP8266-HomeKit-IR-Remote_2024-04-09.csv)

closed 2024-04-09 by HomeKidd

zsoltturcsan, 2024-04-09:
Szia, 

Köszönöm! 
Hibakereséshez és programozáshoz szeretnék még kapcsolási rajzot is kérni. 

Üdvözlettel: 

Turcsán Zsolt 

-----

HomeKidd, 2024-04-11:
[Schematic_ESP8266-HomeKit-IR-Remote_2024-04-11.pdf](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/files/14944458/Schematic_ESP8266-HomeKit-IR-Remote_2024-04-11.pdf)